polyend/TrackerBetaTesting #933: [1.3.1] Sample Recorder - Changing gain (7 and above) whilst monitoring Radio removes sound from left channel

tinysynths, 2021-03-27:
## Hi, sorry you had trouble!

### Steps to reproduce

Sample Recorder - Set Source to Radio. Monitor set to On. Adjust Gain to anything 7 or above and the audio completely disappears from the left channel and is only heard in the right channel from stereo output.

### Other important information

* How often does this bug happen? 
Every time.

* Does this bug happen in previous firmware versions?
Not that I recall.

* Does this bug happen in every project?
Yes

tinysynths, 2021-03-27:
Similar reports already in #912 and #812 

To minimise crossover in bug reporting, please feel free to close.

Jacknametrouble, 2021-12-31:
Hey @MortalJS @ambv @ziej8asek any comments on this one and the similar #912 and #812 ?

closed 2022-07-14 by MortalJS