polyend/TrackerBetaTesting #938: [1.3.1] Live recording bug - unwritable track

keith-macleod, 2021-03-31:
I've encountered a strange behavior while using live record.

The situation can arise where live record will refuse to record on a certain track. My impression is that it generally happens after deleting the contents of an already populated track, but the occurrence is intermittent, so there may be other factors.

When recording, it will record the steps on the next track over, or will skip to the next, or the next... if the track is occupied or is in the same state. If record enable is turned off for all but the offending track, nothing will be recorded at all.

visualearner, 2021-04-03:
I've been experiencing the same a whole lot. It happens around 30% of the time after I've begun recording to a newly armed track. An additional behavior I've found is that it will decide not to Live Record to say--- steps 1-15 but then allow live recording once it reaches steps 16-34.

x2db, 2021-04-13:
same here, the live record behaves very strangely.  This was not an issue  in the 1.3.0 can anybody confirm that this started in 1.3.1 ?

keith-macleod, 2021-04-14:
Anecdotally, I didn't encounter it before 1.3.1 either. I suspect this is the latest chapter of trying to wrestle with the nontrivial problem of recording polyphonically over multiple dynamic, independent, and monophonic tracks.

The situation now has notes spilling over to the next available track when the current step is "occupied", which often just means the note off from the previous note is being recorded, instead of the note you're playing "now". This has the nasty side effect of making it impossible to play legato on a single track, you'll always need to edit after the fact to pick up all the notes that strayed onto different tracks.

So, it seems to me that this bug is a side effect of the way the decision about whether a step or track is occupied is made, i.e. there may be a flag set on a track that marks it as not a candidate for recording, which normally would cause the overflow notes to spill over to the next higher track, which in this case is not being properly updated, leaving the entire track unable to record.

Either way, A couple of changes that I'd like to see for live record mode would be:

- Deprioritize "off" steps so that they can always be overwritten/will never force a note to the next track.
- Any note being played "now" will overwrite the "some" note on the currently armed tracks. e.g oldest in current recording, or newest, or lowest track number, the point being that the note that I'm playing "now" is always the most important one. Maybe a behavior for preferences.
- Have micromove works backward as well as forward to resolve some of the note conflicts that result when playing consecutive steps. This is probably more tricky than it sounds.
- Have a "clobber" record mode that will replace everything on the currently armed tracks.
- Undo completely undoes the recording, instead of note by note.
- A config option to choose whether record is armed by default, or disabled by default.

These are of course self-serving wishes, but I think the first couple are minimally necessary to have a monophonic/1 track record mode that is well-behaved, which is the only thing I really wanted in the first place, and I have to think that I'm not alone here.

PITmast, 2021-05-07:
Similar for me.
It's very buggy and annoying. I'm working with midi, mostly.
The most important for me is to keep playing live pattern(s).
During the play I enter Live Recording (pressed and keep REC + PLAY) and record arpeggiators variations in one track with note-off messages (Live Rec is required for them)
Then I want to delete entire track, so (REC twice) INSTRUMENT, mark track, delete. Everything fine so far.
And then, verrry often, after enter LiveRec again there is no response from midi keyboard (no sound, no recorded notes).
I noticed that pads are recorded but instrument is changed to any below 51 (usually it's silent because I use only midi).

Stop and Play pattern helps but this is what I want to avoid the most.
Sometimes after few attempts with keyboard and pads it finally goes back to work.

Dear Ploy, please fix it. Please, please!
Regards

PITmast, 2021-05-08:
Well, I think I've found where is the problem. If entire track was marked and erased during playback, it's still staying occupied and unavailable to record until playback stop.

To record midi arpeggiator I used to arm and record only one track. After erase armed track new record is not possible until stop.

Regards

keith-macleod, 2021-05-08:
Sounds like a different problem. When I've experienced this, the track remained unrecordable until I restarted or opened a different project.

adamvlasak, 2021-09-01:
Firmware: 1.3.1

Steps to reproduce:

1. arm first track
2. start live recording
3. delete that using select all notes and delete button
4. repeat step 2 & 3 at least 3 times
5. Notices live recording no longer puts your notes

Expected behavior: live recording still works regardless on how many times you delete your recordings

Happens even without external MIDI controller.

adamvlasak, 2021-09-01:
Looks like this works OK on 1.4.0.

MortalJS, 2022-02-22:
Duplicate #469. |Closed

closed 2022-02-22 by MortalJS