polyend/TrackerBetaTesting #940: Wavetables are fun, but they could be funner

JoeDobise, 2021-04-01:
Obsessed with wavetables on the PeT, loaded up a bunch I have and it's been unbelievably inspiring for me.

I have been unsuccessfully trying to create wavetables directly on the Tracker, however. While I have gotten close to useable results, it's very difficult to catch a frame with any precision without using a computer. 
One possible solution would be a wavetable frame display during recording and prior to help tune an oscillator to the correct pitch. 
Another method that could be very useful would be interpolation between single cycle waveforms to build out wavetables of varying sizes. 

Some method to complete this on the tracker itself I think would be a killer feature.

theorize999, 2021-04-09:
This sounds like it could be a good opportunity to enhance the audio editor, I could imagine a couple ways that could play out with an "append" kind of function and a function that blends a selection, say for instance it locked selection size to 4096 samples and then the algorithm would parse that selection into two blocks of 2048 and then make an "in between" and insert it in the proper spot in the selection. I would like to see the audio editor see some more love in general also. I'm guessing improving the sample editor is the easiest way to implement some kind of wavetable builder because it wouldn't involve creating a whole new section of the tracker, the hard part (my guess at least) with making big changes is making it work in the interface.

JoeDobise, 2021-04-09:
> This sounds like it could be a good opportunity to enhance the audio editor, I could imagine a couple ways that could play out with an "append" kind of function and a function that blends a [...]

Absolutely agree. I mean all of it can be prepared in advance then brought over really easily, the workflow with a computer is quite great. I would much rather just say, tune an oscillator to a specific pitch, and record say a filter tweak then turn that recording into a wavetable on the PeT. Been thinking about this, at a proper tempo it may be possible to get some precise wavetables built from single cycle waveforms sequenced on the tracker. Maybe add some effects. I think at 44.1 2048 is 46 ms, (2048/44100=0.046s) a 16th note is 46ms at 326bpm (60000ms/326bpm=184ms per quarter note, 184/4=46ms). So potentially at 326bpm a single cycle wave form (@2048) on each step could be rendered from a pattern to become a wavetable. I might be totally off here though with my math, going to experiment with this soon.

theorize999, 2021-04-09:
I like the way you're thinking that's clever. I don't disagree with the original request though it would be cool, something real nice and easy would be cool and the develpment team does some clever things maybe they can see a path to making something more specialized, my imagination just isn't seeing how it could be done easily. I've been making a lot of wavetables on the computer and just migrating them to the tracker, I like doing it that way because I can use several different approaches and several different tools. 
Possibly the best thing would be for the tracker to just make a wavetable out of whatever sample you feed it, you just define the window size and it does the rest, kind of how a lot of vst's do it behind the scenes it will build a whole new wavetable out of the sample you give it. For instance if you give wavetable VST "X" a sample that has 4 slices in succession that are 2056 samples in length, with the audio file length being 8224 samples in length. Internally the synth is going to have some format it works in, like it prefers, 256 frames with each at a length of 1024 samples, so it's going to parse out the individual waves and interpolate the in between frames and downsample it... which is why some wavetables have loading times, it's converting all this in the background. At least that;s how I understand it, I could be way off it wouldn't be the first time. 
Anyway my whole point is having more functions for the sample editor to append a sample and add in single cycles and then running another operation similar to what I described above would be a pretty darn clean way to make wavetables on PT if you ask me and might be the easiest way to implement it also. At least for the on board way, as far as doing it for incoming audio I'm not sure, Only way I know is trial and error, but would be interesting if someone figured out a way to monitor incoming audio for those kinds of purposes, I suppose if the wave was simple enough it could do some kind of analysis based on zero crossings, you could always calibrate your synth with a simple wave so the tracker could count zero crossings and do a calculation to get the single cycle length and then you tune your synth until it reaches some power of 2, like 512, 1024, 2056.... and then you adjust your synth from that point to have more complex waveshapes and you're free to sample a wavetable of however many frames you want? Am I reading that right , for the recording method you are wanting to sample a HW synth to make wavetables?

theorize999, 2021-04-09:
it's kind of funny I started with "I can't imagine how it could be done" and then as I typed I figured out a way to do it. Kind of funny. Well I'm not sure if those kind of processes work for you but that seems easy enough for me.