polyend/TrackerBetaTesting #941: Random song generator

tfspsound, 2021-04-01:
How about a function to randomly populate the sample loader with random samples, which then get randomly distributed into the pattern to create random songs? Every time I hit the button a random hit song is generated.

benc-uk, 2021-04-01:
This sounds like a terrible idea.

Random samples would just be a mess, unless it was a very limited set of sounds that work well together. Then putting them into a random pattern would sound even worse.

I don't know why you'd waste polyend's time with this sort of thing

tfspsound, 2021-04-01:
Nah man. It could have a section selection device ‘work well together’ and you could set it to -50 to +50 to deal with those issues easy peasy.

benc-uk, 2021-04-01:
Spoken like a true software developer

tfspsound, 2021-04-01:
Ok so I’m not a software developer but it’s the idea that counts. 
Imagine the amount of time we could save instead of crafting everything precisely how we wanted it we could just hit a button and have the PT do it all for us, don’t like it? Hit that button again. Even if it took 5 mins to process and randomly generate a song it’d be worth it to save all that time. They could even add a genre button.

bambrose, 2021-04-01:
Honestly. I don't like being negative or just dismissing things for no good reason... 
but there are already functions for randomising samples (slices too) and i personally think that's enough.
Random sample placement will just sound diabolical and it's very easy to create random variation with careful use of the fill function. That then gives you the degree of control necessary to make it work. It does seem like development time might be better spent in other areas to me. No harm in putting forward the suggestion however.

tfspsound, 2021-04-01:
Hey bam thanks for the opinion and I agree with you 100% about both expressing opinions and PT functions, but you’ve gotta admit. Random is cool right? And just imagine the possibilities, hit random, generates random song. Do I like it? Errrm yes. File, Save. Do it 5 more times n I got an EP from 18 button presses. Talk about workflow. It’s still totally creative because I choose what I think is cool and what isnt. 
Total game changer.

tfspsound, 2021-04-02:
Thanks to those who played along. This was a classic case of April fools.

bambrose, 2021-04-02:
GitHub lolz

benc-uk, 2021-04-02:
You got me.

There was a non April fools suggestion for this feature almost exactly the same, a while back.

closed 2021-04-02 by Jacknametrouble

tfspsound, 2021-04-02:
Yeah soz guys. Just glad u were good sports about it. If we hadn’t been on lockdown I probably wouldn’t have bothered. 

Doesn’t surprise me this has been suggested already either.