polyend/TrackerBetaTesting #944: [v1.3.1] Volume Fx Clicking

Unsettled-Sound, 2021-04-02:
Hi,

I have been trying to use the volume effect on FX 1. Everytime I do this I get a clicking sound from the track. If I remove the effect the clicking stops. The clicks are not consistent. Sometimes it is one of two per loop cycle, sometimes more. The clicks are always there though. 

I am new to the Tracker so this is the first firmware version I have tried (v1.3.1).

trutypesounds, 2021-09-23:
It's still in 1.4.0, on low values volume fx makes sample sound kind of super snappier/clickier (the lower the value the more it snaps/clicks) like its not a volume down function but a hard compressor function @ambv

KezzieCoyote, 2021-11-28:
This issue appears on firmware version 1.4.0 when automating volume parameters during sample playback, it will create a clicking sound as it instantaneously jumps to the next value. A simple fix could be a millisecond or two envelope while it transitions to avoid this popping noise.

Here's a video showcasing and then describing the issue:
https://youtu.be/Y_PFj4bN8XM

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS

MaxDesiatov, 2022-04-11:
This still occurs for me with firmware 1.5.0

faulex, 2022-04-12:
> This still occurs for me with firmware 1.5.0, for both Volume and Low-Pass Cutoff FX.

I'm also having this Issue on 1.5.0

madwreck, 2022-04-15:
I am experiencing this issue with volume and lp cutoff fx as well - firmware 1.5

trutypesounds, 2022-05-31:
its still in 1.6 betas @MortalJS