polyend/TrackerBetaTesting #946: Simultaneous Multiple Patterns Feature Request

clickysteve, 2021-04-02:
Hey there,

I am requesting a change to the workflow to make combining different 'tracks' within a pattern easier.

Currently, if you want to have multiple different phrases from the same instrument which play over a single drum beat, you need to duplicate the drums onto different patterns, and change the melody. 

In other words, if we have our drum beat in Track 1 of Pattern 1, and a lead line in Track 2 of Pattern 1, but want to have a variation on that lead melody, you need to duplicate the drums from Track 1 of Pattern 1 to Track 1 of Pattern 2, and change the melody accordingly. 

This works fine, but means there is a lot of unnecessary duplication with patterns, and it becomes much trickier to remember which patterns have which drum beat (for example).

In performance mode, one way to address this would be to separate out instruments into different Patterns, so drums on Patterns 1-10, bass on Patterns 11-20, etc. This makes it much simpler to manage, and allows for quicker swapping of parts - but also ends up with a bunch of wasted tracks, since a bass line is unlikely to use up all 8 tracks of a Pattern. It also means that you can't really use song mode, because you can't play Patterns simultaneously.

The ability to swap out different parts is made easier with the workflow of trackers such as LSDJ, where the main screen is the 'song' mode, with four parts that can be played simultaneously. 

I am aware that there will be CPU limitations to prevent multiple songs playing at once (since 8 tracker per pattern would add up quickly), but I would love to see an enhancement to allow individual parts (such as collections of instruments) to be more easily switched out, and not have to duplicate them across patterns. 

Thanks!

pinkcell-niico, 2021-04-02:
This is the workflow of LSDJ but it is an ovni in the tracker world since tracker usually do not rely on phrases structure. It surely looks a waste in term of storing repeated structure and in copy-paste. I would say it's the core structure of the tracker that you are on. But the way the Performance mode dissociates tracks from patterns goes your way. :)

dimatura, 2021-04-07:
A more flexible song mode would indeed be amazing. LSDJ (and related trackers like LGPT and the M8) isn't the only tracker that has something like this, but it is probably one of the more elaborate in that it has the three level structure with patterns-chains-phrases.
I don't think something exactly like LSDJ would work for the polyend, but something simpler, probably a simplified version of what Buzz tracker or Klystracker do would be great. Basically allowing each track in a given slot to come from a different pattern. As you said the performance mode already does this, so I suspect it's feasible CPU-wise, and it can probably be done without major changes to the current workflow (unlike LSDJ-style sequencing).

dimatura, 2021-04-07:
Also related issues: #355 #753 #805