polyend/TrackerBetaTesting #947: Simple UI Fix: New Project Confirmation

clickysteve, 2021-04-02:
Hi there,

A simple one. Currently there is no confirmation when you create a new project that one has been created. This means you can sit and wait for ages like an idiot, or needlessly create new patterns. A simple popup to say the new project has been created would make for a more intuitive workflow.

NoraCodes, 2021-04-03:
Seconding this. The `File` UI is pretty sparse, and even just popping up a big green checkmark - "New Project Created" - for a second or two would be lovely, or maybe you could bounce back to the pattern when a new project is created.

pinkcell-niico, 2021-04-05:
I was about to feedback on this and suggest to take us to the Song screen once blank project is created.
1) We do not need to stay on File Screen (unless saving as but since project is always saved, no risk to forget to do this later)
2) The Song Screen shows used patterns (if used in song) and used tracks in white : so a blank song will be readable to figure out that we're on a blank project as expected (although a feedback message shall help too)

Thanks :)

closed 2021-12-02 by Jacknametrouble

NoraCodes, 2021-12-02:
Hi,

Is this problem solved, or are you just closing every old issue?

Nora

bambrose, 2021-12-02:
Judging by the two hundred closed cases (in five minutes) they’re not
bothering to read/assess any j; ust closing all open cases