polyend/TrackerBetaTesting #950: Scroll with jog in Rec Mode

pinkcell-niico, 2021-04-02:
Hi !

I totally love the way we can transpose / increment / decrement values en Rec mode thanks to the jog.

But it hurts the so natural and so efficient scroll-the-pattern behavior of the Jog when not in Rec mode...
Arrow keys are nice but really far slower than this intuitive scrolling.

On classical trackers we could move quicker with Page Up/Down keys (I saw a request here about that) but the jog does this so well (fine or quick scrolling).

My main concern overcomes when I want to edit and move smoothly in pattern while listening to that pattern : I could quit Rec mode to scroll with the job. But.. In that everyday case : the pattern scrolls itself since we're in playmode.... :(
(so my cursor goes wandering around :) until I stop playing or come back to Rec mode)

Couldn't we imagine a key combination with the jog to preserve the current behavior :
Let's say a Screen key or the Rec key itself + Jog.
Furthermore, it would prevent us from entering and changing values by accident (happens all the time).
And the jog would always scroll by default in any mode.

We could think the opposite (key combination to scroll in Rec mode) but I would consider more consistent UI
1) to prevent accidents
and 2) to have the jog scrolling as it does everywhere else in the UI.

Am I the only one to enter/change values by mistakes and wanting to scroll in Rec Mode ? :)

keith-macleod, 2021-04-03:
I'm with you in that I find the arrows to be very cumbersome for scrolling on long patterns. I'd be equally happy if there was a shift+jog combo for scrolling.

NoraCodes, 2021-04-03:
I'd love shift+jog; I don't think changing the default behavior is a good idea.

pinkcell-niico, 2021-04-03:
Shift-jog is also used to translate step octave by octave instead of nite by note. :)

keith-macleod, 2021-04-03:
I know one of these things that I would gladly trade for the other.

x2db, 2021-04-13:
> Shift-jog is also used to translate step octave by octave instead of nite by note. :)

Yes in record mode (red) but in green the shift+JOG would work right ?

pinkcell-niico, 2021-04-14:
In green-mode, while music is playing, the screen auto-scrolls. Making it impossible to move with the jog as long as music is playing.. Too bad.

Maybe we could come to a compromise with a toggle option "CURSOR FOLLOWS PLAYHEAD YES/NO" as you can find in DAWs or impulse tracker (at least). That way, we do not change currents habits. Scroll in green-mode is more consistent. And we get a way to jog-scroll in music-is-playing-mode. 

Note : That toggle option would also be helpfull when you want to hear and work on a specific set of row sand hear it looping. That's awfull when you select a zone, then want to use PREVIEW but cannot hear in loop  and edit since the selection of the zone fades away as soon as you move cursor. There's something to dig here too.

closed 2021-12-02 by Jacknametrouble