polyend/TrackerBetaTesting #951: v1.3.1 Screen freeze in Config menu

musicman799, 2021-04-02:
## Hi, sorry you had trouble!

### Steps to reproduce
Polyned  Tracker plugged into electrical outlet.   Turn on Tracker.
Go into Config menu. 
Go into MIDI menu
Arrow down to Notes In  press Left arrow
Turn Mod wheel or use arrows,....screen jumps to USB ( USB is highlighted)
Screen freezes with no response from any keys.  Power button unresponsive.

I must unplug power , then replug power to restart. 

I discovered this by accident, thinking that I was setting up Tracker to receive midi notes from Sequential OB-6 to program a pattern, while live recording.   I assumed I must set Tracker to MIDI In jack to receive the notes......that is why I tried this.

Tracker will sequence external gear like ASM Hydrasynth and the OB-6 via midi out , with NO problem. 

This tiny glitch does not interrupt play.   It does not seem to affect any other function.

Filmed a video,...but ,
Sorry couldn't get Dropbox to work.......never can ...always have problems.  I quit when 10 minutes of my time has expired.

galenwp, 2021-04-04:
I'm having the same issue. 

Did you figure out how to get MIDI in to the tracker?

galenwp, 2021-04-04:
I managed to get around this by doing Firmware > Reset Config. After a reset everything works normally.

musicman799, 2021-04-06:
I will try this . 
Ive been watching tutorials.  Trying to figure it out.

Thank you

musicman799, 2021-04-06:
Yes . That worked.   Now Tracker is accepting MIDI in.  And screen does NOT freeze when I go to Note In menu under Config menu.

Thanks ,  You solved it .

musicman799, 2021-04-06:
Galenwp  has solved it. ! 

It works.  I reset Config.   Then when I go back into Config > MIDI > Note in , it accepts the change and the controls ( buttons and screen ) do NOT freeze anymore.   Now Tracker accepts incoming MIDI.    Things seem to work normally.

Still learning this device.  

Thanks.

closed 2021-04-06 by musicman799