polyend/TrackerBetaTesting #952: Scroll ALL tracks at once

MattWilsh, 2021-04-03:
It would be great that when you use the arrow keys to move across tracks to have a setting that would allow you to scroll to the next 4 tracks rather than 1 at a time.

E.g. You're on track 4, you can see Tracks 1-4. Press arrow key to right from track 4, you can see Tracks 5-8.

The current way of scrolling is fine but this would be a great improvement to workflow for some.

x2db, 2021-04-13:
Well, shift+arrow is used for selection but double click the left or right arrow could do this. Thanks for this suggestion really useful one !

MattWilsh, 2021-04-14:
Great to hear I’m not the only one! 

I was hoping that it could be a toggle option in Config so something like ‘Normal Scroll’ and ‘Alternative Scroll’ rather than a double tap. That said, you’re is a great idea too.

closed 2021-12-02 by Jacknametrouble