polyend/TrackerBetaTesting #953: Allow deleting/resetting individual instruments from the project.

NoraCodes, 2021-04-03:
> Before you start, did you check whether your feature was already requested?

I did! As far as I can tell, it hasn't been requested yet; the closest thing is #860 and the issues I linked there, but this is more about the project than the sample library.

> Please tell us about your feature in as much detail as you can provide.

I would really like to be able to delete/reset individual instruments from the project, rather than just Delete Unused. I imagine this might succeed instantly if the sample is unused or come up with a dialogue - "This instrument is used in a pattern, are you sure you wish to delete it?" - if not, and would totally reset the instrument so I could start fresh.

pinkcell-niico, 2021-04-03:
I think that what you’re asking for  may already exist. Go to Sample Loader > select the sample in the right list (your song) > press the Delete/Backspace key :)

NoraCodes, 2021-04-03:
Oh, neat! It would be nice to document this somewhere - is it in the manual and I missed it? I ctrl-F'ed for `Delete` when I first looked a the manual.

pinkcell-niico, 2021-04-03:
I don’t think that’s in the manual I just finished reading too :)
I think I noticed it in a video not dedicated to this action ^^

closed 2021-12-02 by Jacknametrouble