polyend/TrackerBetaTesting #954: Mutable Instruments Plaits synthesis modes

NoraCodes, 2021-04-03:
This is kind of a silly one, perhaps, but: Tracker already uses the Mi-Verb, so there's clearly not a license issue with that code. I know Tracker is intended to be a sampling powerhouse more than anything, but it would be extremely cool if we could have some Plaits or Braids modes available for synthesis!

The UI for these would not, in my opinion, be so hard to make; each mode has only 3 parameters, so you could easily have 3 FX options for "timbre", "morph", and "harmonics", plus maybe one for "AUX crossfade" if you want to implement that.

Again, I recognize that this is a long shot, and "this would take too much dev time to consider" or "there is not enough spare CPU time to do this consistently" are absolutely good answers, I just think it would be cool if it's possible, and it would take Tracker to a whole new level.

Thanks so much for making such a wonderful instrument.

theorize999, 2021-04-04:
I wouldn't say no to this. I'm guessing you nailed it with the whole can the CPU handle it thing though, but never hurts to ask and it would be pretty awesome.

charliekemp, 2021-04-14:
This would be amazing if possible. I've been using Braids in M8 Headless and it works so well with this style of sequencing.

Jacknametrouble, 2021-12-18:
Thank you for sharing your idea. This won't be possible due to the limited resources of the instrument.

closed 2021-12-18 by Jacknametrouble