polyend/TrackerBetaTesting #956: 1.3.1 Unresponsive Grid Pads + Fast Playing not possible

visualearner, 2021-04-03:
Hello,

I'm unclear if this is a bug, something I've missed when reading through the manual or a hardware issue but:

When attempting to play the grid pads at a fairly fast tempo only half of my
note pressed will trigger sound. Some notes will light up but zero sound will play. Sometimes its related to another
note being held by another finger and sometimes maybe not. The result is really wonky and forces me to play very slowly. 

At times notes will sound a bit delayed from when I have pressed the pad. When playing fairly slowly they kind of work but there are still times when it seems they will only trigger notes if I hit them really hard. If I don't give them a solid hit, nothing sounds. 

When Record+Play is allowing me to enter notes during playback sometimes
I get an "OFF" slot recorded in between every single step even if I'm
playing the notes fast enough to be entering note information to every
step. 

All this of course makes it incredibly difficult to play in time with other tracks / metronome.

bambrose, 2021-04-03:
Yeah there's a few threads relating to recording. Is linked to note off messages preventing notes being triggered or something I think. Had been noted/ mentioned elsewhere

notaprogrammer9, 2021-04-06:
The little pads are absolute rubbish. 
Not playable at all and they sometimes don’t respond to a finger press. 
But what do you expect for 500€?

tfspsound, 2021-04-08:
I thought it wasn’t possible to play fast but then I realised using light pressure the pads seem to respond a lot quicker. The pads, on mine at least, seem very sensitive so it’s difficult to gauge the correct pressure so it’s still not 100% effective but it’s better.

visualearner, 2021-04-08:
> I thought it wasn’t possible to play fast but then I realised using light pressure the pads seem to respond a lot quicker. The pads, on mine at least, seem very sensitive so it’s difficult to gauge the correct pressure so it’s still not 100% effective but it’s better.

That is very strange. After a bunch more time on it I've had to change my playing style to something I'm not used to: If I tap the keys quickly they don't trigger(!) but if I "press" as apposed to "tap" they do trigger... It's really strange behavior to get used to. I'll have to try light and fast style to see how it goes. 

I'm now finding that tapping the pads at a slow tempo but with a little force gives me only 2 out of 5 triggers...

x2db, 2021-04-13:
> Yeah there's a few threads relating to recording. Is linked to note off messages preventing notes being triggered or something I think. Had been noted/ mentioned elsewhere

So basically any OFF on the track prevents live recording to work properly. Have you noticed this in earlier versions like 1.3.0 ? Also if you could link the other thread related to live recording, thanks !

musojon74, 2021-04-14:
Yeah I raised this too a while back when I first got it. I can hammer notes in fast on an Amiga but on pt I have to play real slow.

MortalJS, 2022-03-19:
No response. Closed.

closed 2022-03-19 by MortalJS