polyend/TrackerBetaTesting #960: [1.3.1] Can't render/export when PT is MIDI slaved

anemlife, 2021-04-04:
Render/export modes don't work if the PT is slaved (Probably related to the missing clock info when trying to export).

I think the PT should "remember" the last BPM used when doing export/renderings. 

Another workaround could be to let us change the internal bpm of the PT even when it's slaved (so the pattern/songs would be synced to the external clock but everything related to export/renders would be synced to the internal clock that we set)

Hope this is understandable
Thanks

MortalJS, 2022-02-01:
https://polyend.com/manuals/tracker/#audio-rendering

The MIDI Clock has to be set to Internal in order to render the stems/master tracks.

closed 2022-02-01 by MortalJS

anemlife, 2022-02-02:
@MortalJS But couldn't the PT automatically use the internal clock (the one that was set for the project before using external clock) when rendering? Having to manually change the clock source to render something seems pretty long for such a simple task, don't you think?