polyend/TrackerBetaTesting #962: Sort File Navigator Alphanumerically

acerier, 2021-04-05:
## Hi, we're excited to hear your idea!

I searched, but could not find this feature in the manual (v 1.3.0) or in this issues database.

### A new feature

The sample projects included with the Tracker use a filename prefix to identify who created it.  E.g. "`<creator name> - <project name>`".

I do this with my project files too.  E.g.  "`ajc - my song`".

If I could sort the file list in the file navigator alphanumerically, then it would be easier to find a particular file, and all my files would be sorted together.

ketumatek, 2021-04-05:
Or/and add the date the projet was create...could be something nice too ...

Just-Peter, 2021-04-06:
> Or/and add the date the projet was create...could be something nice too ...

Date is not possible, the Tracker has no internal clock. So it has no clue of date and time.

x2db, 2021-04-15:
> > Or/and add the date the projet was create...could be something nice too ...
> 
> Date is not possible, the Tracker has no internal clock. So it has no clue of date and time.

no problem, just have internal counter that increments any time you save a file, then store this number with each project/file and you can sort it this way, no need for date