polyend/TrackerBetaTesting #964: Filter slide commands

theorize999, 2021-04-09:
I'm sorry if this has already been requested, I couldn't find it if so, at least with it's own request, I think I've mentioned it in a comment. Which is kind of baffling, I would think this would be one of the first things requested, but maybe my search query was wrong or something. 

Anyway, I often feel the need to do filter sweeps. I would like to be able to smoothly automate filter cutoff, it's kind of a staple in electronic music at least for me.  Polyend Tracker is sorely missing this feature, and simply setting the cutoff on each step doesn't come close to cutting it, it sounds exactly like it looks. Super steppy. This is usually accomplished in trackers I've used in the past with simple "filter slide up" and "filter slide down" commands, where the variable determines the speed of the slide. This feature can even be used to make more complex envelopes than ADSR, it just takes the position of the filter and slides it from it's current position. It's OK that some filter slides might go against the envelope if you do it in just the right way, most of the time it will only make the modulation contour more interesting. 
I've seen some trackers have a parameter in the instrument or track settings called "inertia" which I can't fully explain but had to do with how quickly filter slides get going in the new trajectory, sort of if the parameter had a kind of weight to it. I don't think that would really be necessary here though I'd be super happy with just plain and simple slides, that way I wouldn't have to make a new instrument and waste sample memory just to do some kind of variation in filter movement.

Thank you for consideration and if anything need clarified on, if I did a bad job explaining what I want please comment on this I'll get the email notification. I see this as hugely important especially considering the fact that the tracker is still being developed, I feel like the existing features that need refined should be addressed before too many new features are implemented, while it's still possible.

theorize999, 2021-04-09:
we could take this idea even further and apply it for example to wavetable position sometimes we don't want just static modulation on wavetable position like envelope provides, same modulation every time. it would be really nice to be able to set it with one command and slide it with another. We can carry over this thinking to the granular position too, not every parameter in the tracker needs to have slide commands but a few key parameters to be able to slide would be nice and I think everyone would agree that it would only improve the tracker especially it's overall sound would be smoother with smoother modulations.

ketumatek, 2021-04-09:
It is a sort of automation you can apply to the filter...nice idea that can be apply to another feature maiby...

theorize999, 2021-04-10:
Yes Pitch slides are pretty standard for trackers too, I was somewhat shocked to see that they didn't include these as FX commands from the beginning, it's always been the main tool in trackers for any kind of expressiveness. Please consider pitcch slides as well Polyend, the tracking community will thank you... you would be surprised how many people still rely on things like pitch slide and arpeggio. I've never done a chiptune with a sample tracker in my life but plenty of people do and if PT had the right commands I would definitely do some... PT has the potential to make some really amazing sounding chiptunes, and I'm really picky when it comes to chiptunes that I like.

math0ne, 2021-04-13:
You can achieve nice filter sweeps using the LFO(set the step length, and saw / rev saw) in combination with the main LFO effect.  It's a little finicky but you cab get pretty good results.

theorize999, 2021-04-14:
yes there are some excellent modulations that can be done with the LFO but it's in no way a replacement for a proper slide command.... number one you need to give up the envelope and that ruins half of the sounds I'd want to use slide commands on right off the bat. Plus while messing with the LFOs is great but if you push it too much it can kind of glitch out the sound, That can sound cool at times but not always desirable. With wavetable and filter slides you could easily do acid lines in a couple different ways, of course that's not all you can do but a pretty good illustration of something that's automation heavy. There's no denying the expressive potential having slide commands inject into a tracker.

math0ne, 2021-04-15:
Yeah for sure, I would guess we would still have whatever glitchyness no matter how filter sweeps are done.

I do want to push back slightly on the LFO because I don't think you are correct about giving up the envelope.  In a column with an instrument that has an envelope you can enable the filter LFO for a few steps then turn it back off before the release with no effect on the envelope as far as I can tell.

theorize999, 2021-04-28:
I thought I tried that before, I'll try it out again to see I hope you are right. If I can get the same effect with LFO then sure! I'll do that, but if it doesn't steal the modulation focus then I need to figure out why I'm getting clipping in Volume Modulation in another module. I've been super busy I haven't had a chance to revisit this until now. I sure do hope you're right if so I can think of all kinds of stuff I want to track when I get another spare minute.

closed 2021-12-02 by Jacknametrouble