polyend/TrackerBetaTesting #966: Save The JogWheel. Have values scroll from 100 back to 1 again using direction keys

bambrose, 2021-04-09:
It's a simple one to describe. 
If you want to avoid rapid scrolling on the jog wheel you can use the arrow keys.. but to get from 100 to 1 takes a long time (and vice versa)

If the arrow keys followed the same behavior as they do when scrolling through a pattern... that is if you go 'back' from 1 you get to 100, and if you go 'forward' from 100 you get to 1... ON ALL PARAMETERS, it would enhance navigating while using less jog wheel.

ConnorCarbon, 2021-04-09:
I agree but I think what you're referring to is vertical wrap on pattern screen, right?

bambrose, 2021-04-09:
Thanks yeah I couldn't think of the correct expression to use.
I'm suggesting that same wrapping would work on most, if not all, parameters on all screens

tfspsound, 2021-04-10:
‘Save the JogWheel’
Good T-shirt slogan if ever I heard one. 
#776 had a bit of support but got closed so there might be an earlier thread for this request too.

tfspsound, 2021-04-10:
Soz that was in related

closed 2021-12-02 by Jacknametrouble