polyend/TrackerBetaTesting #968: [1•3•1] new feature - fx resquest

ketumatek, 2021-04-11:
At this time we have 2 fx colon.
When I select my first Fx in the first colon and I want to put the same Fx in this colon it is always the Fx I chooses.
And if want to take another Fx in the second colon apply it and decide to apply the first Fx in the first colon I have to re select the old Fx.
What I would like it is to fix the Fx in each colon...
I explain for example 

1Fx 1 colon - roll That I apply to sample
After I want to apply volume so I go to Fx 2 in the second fix colon and apply the volume.

But if I want to apply in another place the fix 1 in the colon 1 I have to select again the roll fx

What it should be nice it is when I select the Fx 1 and the Fx 2 it stay like this until I want to change one of them.
It is better for the workflow and like this you don't have to select each time the Fx further more if you use the same 2 fx...

Hope it make sense

notaprogrammer9, 2021-04-11:
That would a nice option to have. 

Together with (also optionally):

- fixed instrument slot for each track;
- fixed MIDI channel for each track

x2db, 2021-04-13:
Excellent feature to have a "sticky" FX selection. When I choose Roll or Chance FX on column 1 or 2,  I usually want to continue with the same effect per column, mixing effects on both columns just puts more confusion to the process. So it would be great not needed to choose the FX anytime I jump away from FX slots :)

keith-macleod, 2021-04-13:
It seems to me that there is a certain amount of stickiness already in the interface, but it's not always clear what it's doing, or that it's entirely consistent.
I would love if there were just a little corner of screen space reserved to display the currently selected instrument and effect, so that I'm less likely to dump a dozen horn samples on my hihat track.

pinkcell-niico, 2021-04-14:
Maybe, just a "BUTTON-COPY" press on a previous item could also recall that spot's intrument, fx1, fx2 ?
Locking something to a track does not seem Tracker-spirit ;)

theorize999, 2021-04-14:
Great idea