polyend/TrackerBetaTesting #969: [1.3.0] USB transfer of WAV files freezes/crashes

notaprogrammer9, 2021-04-11:
Hello

I try to use the Android Transfer application on osx 10.15

Some issues:

issue #1 
When creating a subfolder, the folder gets placed on the root of the card instead

issue #2
The app randomly crashes (freezes) when dragging multiple files (I tried 5 WAV files) at once. No crashes when dragging one by one.

I’m aware this is a third party app and out of reach for Polyend. But is there any robust way of transferring-organizing the card over USB? 
Like regular USB storage mode...?

I’m using the original polyend USB C cable and a 2018 mac mini

pinkcell-niico, 2021-04-12:
Oh damned yes. The only thing I could came with this Android File Transfer app (between constant and untimely disconnections) is : creating a folder with default name, unrenamable, une moveable and I have never been able to copy or move any file.. (I'm on OS 10.14
If you could come to a USB Mass Storage behavior or similar, this would be great to organize things :)

notaprogrammer9, 2021-04-12:
This third party app shouldn’t be linked to on the official Polyend website. 
It doesn’t function on Mac.