polyend/TrackerBetaTesting #970: Use of internal transport while also Synced externally

hoggy4, 2021-04-11:
Im not sure whether to call this a bug or a feature request but I wish I could use the trackers transport while still being sync'd to an external transport/clock as is the case with most equipment.

anemlife, 2021-04-12:
Agree! Did a similar request a while ago ( #831 )

eturnigy, 2021-04-13:
i really hope this is a priority.

sunfallz, 2021-04-15:
Did a somewhat similar request a while ago regarding ext clk and tempo fx (which you currently can't use when ext clk)
Tempo fx only work when using the internal clock.
Example: I have 2 trackers, 1 is the master clock, on the 2nd tracker receiving clock it would be great to have tempo fx work in this mode. (not just config divisions, which would be rad as well), but being able to dial in whatever tempo fx desired per step would be amazing.
#717