polyend/TrackerBetaTesting #971: Upgrading to 1.3.1

theorize999, 2021-04-12:
I'm having problems upgrading to 1.3.1
It says update failed
it boots up and works just fine with 1.3.0 or anything earlier
I can switch to earlier versions of the firmware just not 1.3.1

theorize999, 2021-04-28:
I think this is due to having a new 256 gig SD card, I think 128 might be the limit, I'm going to try flashing it directly and if that doesn't work I'll have to go down to 128 I know that works. I was trying to go overboard with the samples anyway I don't think I need my entire sample collection on there, you can probably close this, it I don't update this by 4/29 please close this ticket thank you dev team, you are awesome :)

theorize999, 2021-04-28:
I did the emergency flash and it worked fine, I was hoping to not have to do it that way, next update we will see how it goes, maybe the problem was changing from 1.3.0 specifically but I'm pretty sure that was the first time flashing new firmware since I got the 156GB SD card... I'm confident that was the cause. Please close.

JamesTheHacker, 2021-05-02:
I have a 256GB SD card and it works fine. You need to format it to Fat32, but Windows will not allow you to do this on a large device. Instead, download rufus: https://rufus.ie/en_US/ and format it to Fat32. Works a charm.

theorize999, 2021-05-02:
thanks I've bookmarked that in case I ever decide to go through this again... I got it formatted to FAT32, it's pretty clear from Polyend's instructions it needed to be FAT32, but whatever it was I used really did not make it easy. If anything I'm just going to go down to 128, I'm not going to use even 10% of what's on there now I pretty much just threw my whole collection of normal samples on there, and honestly I'm making pretty much all my own samples these days except for some acoustic drums and orchestral type stuff. My big thing is I don't keep all my samples on the computer hard drive, so loading anything up with samples is a pain, not something I'm willing to do often. Even if this is some kind of bug I won't know I'm not planning on loading up that many samples again.

closed 2022-01-13 by MortalJS