polyend/TrackerBetaTesting #975: OPL 3

theorize999, 2021-04-14:
I know it's a long shot, but it would be pretty awesome if there was an OPL 3 emulator we could use instead of samples. I'm pretty sure that would also attract the attention of a lot of old school tracker users and would probably result in some really good chip tunes being made. And for everybody who's not into chiptunes well it's still an FM synth that's very usable, could have some real nice FM bass!

It just so happens that there's a really good emulation, probably the best emulation done so far made by Reality for Reality Adlib Tracker (RAD), and the source code is included with it. 

https://www.3eality.com/productions/reality-adlib-tracker

Jeymadcat, 2021-04-15:
or SID emulator : )

theorize999, 2021-04-17:
or both! You just reminded me of a part I need to order for my poor broken C64, I got MSSIAH and haven't even been able to use it joystick port is busted :(