polyend/TrackerBetaTesting #978: Oscilloscope Vumeter for all tracks

Jeymadcat, 2021-04-15:
Hello ! Devs and Customers

Could you consider putting Vumetter-like oscilloscopes like in old trackers on computers. It gives another dimension to the sounds! this is maybe a bit gimmicky and would consume resources. I do not know if it would be possible and easy to make this integration.
this would add a much more organic side to this tracker wonderer  : )
[image]

-

## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature

Please tell us about your feature in as much detail as you can provide.
Vague ideas are less likely to be successful.  Concrete requests are
easier to estimate.

UserOverflow, 2021-04-15:
This was always very important in every (computer) tracker. It contain information about what is happen on track, if it's muted, if it's not to loud, wat is beat of even stage of ADSR. Tracked was created for new users, but also "old one" and this is something what is really missing. Especial in "Performance" section. I'm always confused which track contain beat, bass etc. From this tiny window you may know what is where, even without naming any track.

theorize999, 2021-04-17:
YES!!! I meant to ask for this and forgot! Good job!!!
It's def not a gimmick, it is very helpful, especially to be able to see all scopes at once. Heck whenever I load modules in FT2 the scopes are my main way of analyzing the track. You can see literally everything. Ok I'm exaggerating maybe a bit you can't really know all there is to know by looking at the scopes but you can tell quite a bit.  What would be nice is to be able to rearrange the placement of the scopes, so you can side-by-side compare things. but really l'm asking too much any scopes at all would be great as long as there all on the same page, a dedicated page would be best in my opinion.

k4ot, 2021-09-17:
I second this, for quick visual feedback

closed 2021-12-02 by Jacknametrouble