polyend/TrackerBetaTesting #981: Feature request - SD card file management

pulzev, 2021-04-15:
Anyone would take for granted that the feature of being able to delete and rename sample files from its storage unit would be built into any sampler that is advertised as standalone, but not only is this deal-breaking function not built into the device, its omission isn't even mentioned on the product page or manual. The Tracker would have been a definite no from me had I known it couldn't manage files before I bought it.

I have no issue at all with the other limiting features as they conform to the product's retro theme which appeals to me, but I don't understand why the user can't manage samples when the functionality and keyboard to do so with project files is there. I am mindful of the argument that my proposed feature carries the risk of the user being able to delete the Tracker's firmware folder, but just being able to rename and delete audio files, the only files actually visible in the Sample Loader, rather than folders would be enough for the device to be able to classify itself as 'standalone' in my view. The bottom line is always a PC; If the SD card is full, the user needs to either delete the files on a computer, or buy a new SD card and install the firmware from a computer on to it. I do appreciate that the ability to rename sample recordings by overwriting is available, but do you not agree that this method is an amateur way to go about renaming, and that the usefulness of the in-built sampler would also increase if the user could delete and rename their recordings?

Until the Tracker can delete and rename samples, just like any Akai S series unit from 20-30 years ago can, it will remain an accessory to a computer similar to the MPC Touch instead of the standalone workstation it claims to be. My feature request is therefore to add a way to delete, and rename audio files in the Sample Loader.

anemlife, 2021-04-15:
Been asking for this for a pretty long time. (Requests #860 #342 #582 )

I definitely hope they add this option!

static-noise-PT, 2021-04-20:
yes please at least a delete option

SeanHub, 2021-11-30:
Please add rename and delete file support

closed 2021-12-02 by Jacknametrouble