polyend/TrackerBetaTesting #982: Volume Control for Sample Preview

vverke, 2021-04-20:
It would be nice to have some kind of volume control for Sample Preview (in Sample Loader section). Right now, while previewing samples all of them are blasting at full volume (specially hurtful on headphones when I’m looking for that kicks and snares).

This could be a separate thing in Sample Loader or Config menu. Or maybe it could be linked to Master Volume (so I could temporarily lower the volume there).

static-noise-PT, 2021-04-21:
Another possibility would be to reduce/limit the output in Sample Preview via a General Setting. Something like -6db/-12db/etc.

astrochili, 2023-03-22:
Related to https://github.com/polyend/TrackerBetaTesting/issues/762 and https://github.com/polyend/TrackerBetaTesting/issues/1120 and