polyend/TrackerBetaTesting #983: [1.3.1] Noise while playing back a recording or have it open in Editor

static-noise-PT, 2021-04-20:
As the title says every now and then after recording directly into the Tracker the "Preview" of the sample is super noise-y.

After saving and loading as "Instrument" it's a clean recording without noise. While working on the same project and opening the sample in the "Editor" the noise-artifacts are audible. 

It's not the file itself since the loaded sample in instrument is clean.

It do not always happens. I have the Tracker since 8 days and had not much time to investigate further but this is a bit annoying and quite loud noise-artifacts.

static-noise-PT, 2021-04-23:
I want to add without opening an additional "Issues" that I now ran also into a situation where there was no output at all.
The scenario is as above but instead of noise-artifacts: silence. This was gone after loading the sample into an instrument. So Editor as well as sample preview after recording worked as expected. Is always the first run of recording or have it open in Editor which brings noise-artifacts and now well, silence too

djkurzman, 2021-04-25:
I'm experiencing this as well in 1.3.1. It doesn't happen with all projects, but one I started working on with a specific set of samples gives me heavy distortion when previewing anything in the Sample Editor, but sounds normal everywhere else.

djkurzman, 2021-04-27:
-- I was using the sample recorder quite a bit in that project and noticed the same thing happened in a different project right after using the sample recorder. I went from the sample editor back to the recorder screen, and then back to the editor screen again, distorted playback in the Sample editor was gone.

trutypesounds, 2021-09-26:
had the same issue, it happens random, try to sample line in, try to preview - the sound is noisy. rebooting helps. its still in 1.4.0 @ambv

static-noise-PT, 2022-02-26:
> 

Did not recorded so much with 1.5 but it I also did not ran into this noise since 1.4 I think. Maybe leave it open for a month or so. I have again more time to spend with the Tracker.

static-noise-PT, 2022-03-07:
Over the weekend I recorded a lot with the Tracker. This bug did not showed up again for me. Recording is smooth.

closed 2022-03-07 by static-noise-PT