polyend/TrackerBetaTesting #984: [FR] Set default settings

static-noise-PT, 2021-04-20:
Quick thing hopefully: I would like to have at least an option to set default settings for things like pre/post fader settings, scale, volumes. Default settings by taste would be great.

Lislegaard, 2021-04-20:
Yes please! I have to change my grid offset from 12 to 5 and C4 EVERY time

ketumatek, 2021-04-21:
Looks as a preset template that you can recall for each new track...
Because for the moment you need to make an empty track as you like save it ...and call it for a new projet and save as new projet and then use it...
More simple if you have a template already made as you want that you can use as new tracks...

static-noise-PT, 2021-04-21:
Set default settings is the probably simplest way for the developer. Although a folder named "Templates" which contains projects which are recognized as a new project would be great because multiple templates for different situations would be great.

x2db, 2021-04-26:
... adding to the crowd... I have to do this on each new instrument for ADSR and LFOs 3+3+2 for each instrument, but yes I could create a project called template and load all of these

closed 2021-12-02 by Jacknametrouble