polyend/TrackerBetaTesting #990: Pattern List + Pattern Naming + Split Master/Cue Levels

iodosan, 2021-04-23:
Hi guys!
it would be great to introduce a Pattern list view (as is for instruments and FX when keeping the buttons pressed) alongside pattern custom naming. This could make navigation through patterns and song arranging way faster.

Another thing:
i noticed that when previewing samples in the sample loader the volume can be way louder than what you're playing in your mix out. this is really dangerous when working with headphones. Adding separate Master and Cue/preview volumes in the master view can be really handy and also safer for your ears.

Thanks
Pierfrancesco

ketumatek, 2021-04-24:
For the seconde request add at #982..

arphex, 2022-03-03:
the pattern naming  would be immensly useful and hopefully easy to implement

MarderChip, 2022-12-22:
Naming Patterns would be extreamly helpful for my large Projects.

huggybeats, 2023-01-19:
I really hurt my ears several times, while previewing some samples. the volume control of sample preview is so much needed. Often I have to put my headphones down on the table while previewing. Even having them just hanging around the neck was too loud. But then you cannot properly analyze the sample sound. I hope so much that this will be implemented soon...