polyend/TrackerBetaTesting #991: Suggestion --> More Zoom

anemlife, 2021-04-24:
The zoom "ratio" depends of the current sample length. This is a problem when trying to adjust zero crossing points for long samples (especially when I try to create clean wavetable).

Could the zoom be able to zoom to the same "values" with any sample?

Thanks

tfspsound, 2021-04-25:
Yeah it would be nice if 100 was always the max zoom and it was the same level of zoom across all samples despite length. 
I’d like it if in Config they could add a zoom option so I could set 4 values for zoom that I clicked through (eg 0, 50, 75, 100) then I could just do 4 ‘clicks’ of the jogwheel instead of scrolling. 
Save the jogwheel!

x2db, 2021-04-26:
or **Shift + start/end/loop start/loop end** could automatically pick the closest zero crossing. But I found, that on some samples especially on E-piano (rhodes) or certain bass samples that the zero crossing still leaves me with a click so I need to find a symmetric zero crossing point .. So there is a bit more into an automatic zero crossing sample management. 

my 0.2

theorize999, 2021-04-27:
Often times I still get clicks when looping if just going on zero crossings, I've found the tone of the sound you're looping back to can't be that different unless you're doing extremely short loops like just a few cycles. Can't say as I've ever looped a piano or rhodes sample before but I know they do it extensively in consumer keyboards like Casio's and what not. I think it's a clever approach and I've always wanted to use that technique in a tracker but never got around to it and forgot. Thanks for reminding me! I actually wouldn't mind getting a basic Casio or Yamaha keyboard just to sample and use in the tracker, I'm guessing they're still using techniques like that and I can snag small size instrument samples to use in the tracker. Sounds fun :) 

But yep, I totally agree we should be able to zoom in to sample level in any situation really, I get why it is the way it is, but this tracker has a better screen than that I hope the sample editor continues to get worked on, it's ultimately going to dictate how powerful of a sound design tool PT can be, I would argue that it has loads of potential in this department. I'm guessing a crossfade function (with adjustable % range) would help a lot of looping woes... actually now that I think about it this should be an essential function, considering PT definitely pushes you towards using small looped samples, I like the memory limitation it definitely sparks a lot of creativity for me in making samples to be used in PT. It's pretty fun I'm always on the lookout for sampling material.

tfspsound, 2021-04-28:
@x2db #539 there was some discussion of this if your interested.

closed 2021-12-02 by Jacknametrouble