polyend/TrackerBetaTesting #993: Suggestion - - > Pads from C0 to C10

anemlife, 2021-04-24:
In the config we can set the pads to begin from C1 to C4. The problem is this doesn't cover the whole range of notes! So:

1) could we get a C0 note (tracker goes down to C#0 but by adding the C0 note we could add one layer of pads)

2) in the config we should be able to set the pads from C0 to C7 (so the C7 setup would cover C7/C8/C9/C10 octaves => we could then play every notes with pads)

3) I already asked about this in other threads but I REALLY think there should be a shortcut to change the displayed octaves for pads. 
Going to config everytime you need to change it is way too long for such a basic task. Especially when working with EXT midi devices. 

Thanks

ketumatek, 2021-04-25:
It should be nice if like with shift or something else we could have access to another 4 octave
Like C1 C2 C3 C4
And with shift C5 C6 C7 C8
That is true that it will be easier to use pad and have full range of octave without going in global parameters to change it...

After you have a solution because when you are in pattern mode you can write your note for example C4 and if you want a C6 or C8 you can change it with the scroll ...it is longer but it is work too...

x2db, 2021-04-26:
Shift double tap left bottom pad to go octave down 
Shift double tap top right pad to go octave up

Otherwise to change octave on tracker is 3+12+2= 17 button presses: 
   
Config -> arrow down -> change -> 12x arrow down or up -> accept -> pattern