polyend/TrackerBetaTesting #1000: Note Cut/Off FX

Aisjam, 2021-04-27:
Similar to LSDj's Kill note command. Being able to cut a note shorter than a individual line would allow for some staccato goodness. especially paired with the roll command :).

theorize999, 2021-04-27:
Wow no one has asked for this yet?!?!?! every time I use it I wish for this command, I just figured someone would have asked for this already. PLEASE PLEASE PLEASE PLEASE PLEASE listen to this very wise person and give us this command, I love PT but in all honesty there are a few essential FX commands that pretty much every tracker has that are still missing. This is one of them. I like the idea of pairing it with the roll command that sound so fun and I haven't used any tracker that has both of those actually, well not true many trackers had the retrigger command but the roll in PT takes it further... yes yes now I'm dreaming of tracking goodness.

anemlife, 2021-04-27:
Already asked for something similar in #729 for staccato stuff but your idea to be able to make it shorter than a step takes the idea a step further!

+1

imabigdipper, 2021-04-27:
Looks like the 1.4.0 beta has answered your wishes haha

Aisjam, 2021-04-27:
Yeah looks like. I'll close this one.

closed 2021-04-27 by Aisjam

theorize999, 2021-04-28:
This is AMAZING!!! Best update ever!