polyend/TrackerBetaTesting #1002: Render Selection Freezes 1.3.1

JRW808, 2021-04-27:
Steps to reproduce:
Make a selection in pattern view
Go to Render Selection, add a custom name (with spaces), 
select Render & Load "Exporting pattern" will display and then hang/freeze at the end of the progress bar

I had to unplug the PT to restart it. This was only happening when I had the Midi Clock In and Midi Transport In set to Midi In Jack. But it was happening every time and in every project. It works fine with those Midi setting set to Internal or Off.

JRW808, 2021-04-27:
I just installed the 1.4 beta and am having the same issue. The Render Selection only freezes when Midi Clock in is set to Midi In jack

anemlife, 2021-04-29:
check issue #960 
similar :)

ROM72, 2021-09-09:
Same thing for me too with the 1.4 firmware version.
https://www.dropbox.com/s/by8dslpguvbws4t/Polyend-tracker-freezing-when-render-1.4-firmware-version.mp4?dl=0

MortalJS, 2022-02-01:
https://polyend.com/manuals/tracker/#audio-rendering

The MIDI Clock has to be set to Internal in order to render the stems/master tracks.

closed 2022-02-01 by MortalJS