polyend/TrackerBetaTesting #1003: [1.4.0b1] Bug title

phaelam, 2021-04-27:
## Hi, sorry you had trouble!

### Steps to reproduce

new PERFORMANCE update is SUPER SICK. there is just an issue with the labels

EX: I’m on PATT 3, to test I switch a track to PATT 4 that has nothing. and the track goes silent, but at the end of the loop the label switches back to PATT 3 even tho im still in PATT 4. makes it a little bit of a challenge to get cleanly back to PATT 3.

but other than that…this is SUPER sweet. :+1:t6: :+1:t6: :+1:t6:

_ENHANCEMENT PART: want to add…this may LOOK counter intuitive, or perhaps its how my brain werks with this. but hitting the UP arrow to go numerically up in PATTs makes sense, but not visually. I keep hitting up to go up the list, but I go up numerically [instead of down the list]. 1 sitting vertically above 2…if that makes sense. currently to select PATTs is the opposite of crawling thru the list_

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
** always

* Does this bug happen in previous firmware versions?
* *not certain, been meaning to try this for a time

* Does this bug happen in every project?
* *yes

phaelam, 2021-04-27:
so playin with this a little further, I thinking this isn't a bug, but with leave it labeled that way for know.

what it looks like is the PATT you are looping is displayed at the bottom of each track. makes me forget what they are on. would be SWEET if the loop PATT could be displayed at the top, and the current PATTs you are on display in each track.

tfspsound, 2021-04-28:
Performance mode finally as it was supposed to work. Cheers polyend you guys nailed it. I think this is a bug but not a major bug. Performance mode is working as expected but the text displaying which pattern each track is on is not getting updated. I noticed if I hold Shift and Play (eg on track 1) it will loop track 1 as expected, I can then change pattern and it will loop that pattern but the track text at the bottom of the screen will still read track 1. 
Again, nice one guys, PT just keeps getting better and better.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS