polyend/TrackerBetaTesting #1004: [1.4.0b] Filter LFO on triangle

BerndPeuser, 2021-04-27:
I put a LFO on a Lowpass Filter with 128 steps triangle. During passing a 64 pattern, the Filter starts with 100 amounts in the middle of the pattern playing a loop file. 
Position 01 C-5 Loop 
Position 33 C-5 Loop 

Every 96 steps the LFO starts with 100 amount.

tfspsound, 2021-04-30:
I’m having lfo issues too. Not managed to replicate your issue but it sounds similar. When I set up a pattern eg 64 steps long and add an lfo in sync with a speed of 64, 32, 16 basically a whole multiple, the sound plays through modulated as expected but when it resets it ‘steps’ horribly spoiling the tone. 
Also since envelopes are with lfos I might add that maybe the ADSR needs a little TLC. I’m finding the A and R to my ears sound like exponential curves not linear making gradual changes in sound over time difficult, it seems a bit all or nothing.

MortalJS, 2022-02-22:
Hi @BerndPeuser 
Can you check if the problem occurs in firmware 1.5?

BerndPeuser, 2022-02-23:
Yes. LFO starts everytime on wild positions. I made an example with a string. 
[lfo  96.zip](https://github.com/polyend/TrackerBetaTesting/files/8123742/lfo.96.zip)

MortalJS, 2022-02-23:
@ziej8asek