polyend/TrackerBetaTesting #1005: [1.4.0 Beta 1] Wavetable volume not gradual

MegaTurrican, 2021-04-28:
The Wavetable volume is not changing gradually as it was supposed to do. Simply load a wavetable instrument and turn the instrument's volume all the way down to check it out.

ambv, 2021-05-23:
I don't understand. If you are using a V effect on the track, that overrides the value in Instrument Parameters. This has been like this since 1.0.

MegaTurrican, 2021-05-24:
[test 1-4-0.zip](https://github.com/polyend/TrackerIssues/files/6531928/test.1-4-0.zip

What I am saying is that the instrument volume (and more detectable with wavetables) has lost its gradual resolution and now is more steep. You go from: no sound to loud with a simple change of value. I am not using the volume command. Check out the attached project, track 4, instrument 4, and the attached video. Something has broken the instrument volume resolution and its not as gradual and smooth as it were, now its steep, especially with wavetables.

MegaTurrican, 2021-05-24:
https://user-images.githubusercontent.com/71126897/119340962-e56cd600-bc9b-11eb-9e8b-ab1fcbb623a7.mp4

MegaTurrican, 2021-05-24:
Watch and listen how steep the instrument volume curve is. In immediate previous versions it was much smoother.

bambrose, 2021-05-24:
Yeah I've noticed some strange behaviour around this as well.
The "VCA" in the amp envelope section seems to act independently of the "instrument volume" once you start using rolls like rv and RV

closed 2023-04-26 by MiroPoly