polyend/TrackerBetaTesting #1010: Fill Mode

77377, 2021-04-28:
I'd be quite an improvement if one could combine the Notes/NoNotes/Fx/NoFX option with the every nth step/Euclidean/Random fill option to do something like this: 

Fill (overwrite ) every nth of steps who currently have a note on event
Fill 25% of the steps which currently have no note 
 and so forth.....

Cheers!

tfspsound, 2021-04-30:
There might be a suggestion for this already but I was thinking it would be nice when replacing a sample in a pattern I was given a pop-up with 4 options
1: single - just the cell highlighted. 
2: all track - every cell with same sample in current track. 
3: all song - every cell with same sample in whole song. 
4: all track all song - every cell with same sample in same track in whole song. 
I guess you could add row options too if that helps too like same sample same row, all row all song. It’s make changing out samples a whole lot easier.

77377, 2021-06-11:
Nah not exactly what I talked about but interesting as well, Cheers!

closed 2021-12-02 by Jacknametrouble