polyend/TrackerBetaTesting #1011: 'Favorites' Hotkeys

0x0zzz, 2021-04-28:
My suggestion is to use the 8 black keys beneath the screen as Favorites hotkeys.  The user would have 16 favorites each for Note, Instrument, FX1, and FX2.  Holding its key and then tapping the corresponding black key would select Favorite 1-8, and holding Shift while doing so would select 9-16. 

I would immediately use this to set the MIDI channels to my Instrument Favorites.  Switching between Instruments in the 1-48 bank and the MIDI channels can get a little awkward, especially if you do it frequently.

tomhayes, 2021-05-03:
+1 for this - would also minimise excessive jogwheel scrolling when switching to MIDI instruments etc, which is something I am quite concerned about given people's jogwheel issues.

0x0zzz, 2021-05-03:
I've been cheating that a little by holding the Instrument key, tapping the bottom-right-most pad to jump to slot 48, and then using arrow-down into the channels.

tomhayes, 2021-05-03:
Thanks, that's an awesome tip!

closed 2021-12-02 by Jacknametrouble