polyend/TrackerBetaTesting #1012: "Play Thru"/Continuous Slice Mode

tomhayes, 2021-04-28:
Currently, when we slice a sample, playing back each slice plays from that slice's marker and stops when it hits the marker of the next slice. 

It would be absolutely awesome if we had a mode where slices would play from their respective marker through to the end of the sample (or as far as it can whilst the note is on) instead of stopping when it reaches the next marker. 

This is possible in Renoise with the little 1> button at the top right of the slicing screen, as an example of what I'm talking about.

ketumatek, 2021-04-29:
Good ideas + In Renoise the first slice it use to play the all sample...
That will be nice fracture two..
Because sometimes you need to play the entire sample and for this you need to copy sample in another slot...

gherat, 2021-05-17:
I've asked this a long time ago, still no update on that. It is really lame that the sample stops at the end of the slice.. The sliceponts should be like a start point: "play sample from here". Then you should be able to let it stop by a gate/step time, which plays it back like: [ 1/32th ]    [ 1/16th ]     [ 1/8th ]     [ 1/1 ]     [ 2/1 ] They have the gate time fx now in the new firmware but that only plays the slice shorter :( Like 50%, 25% etc. It should also go up like 200%, 800% etc

dramaone, 2021-06-12:
this... standard requirement well before games

danapeks, 2021-07-19:
+1 on this one, really needed for samplists.
Already made a precious feature request for this and there are a few more. See #461 and #254 etc

closed 2021-12-02 by Jacknametrouble