polyend/TrackerBetaTesting #1017: [1.3.1] note on reset the LFO

PaulAntioche, 2021-04-30:
I don't know if this is a bug but it is a very big problem.
In "granular" mode with an LFO on "wavetable position" when there is a "note on" the sample is not played on the actual LFO position but on the position setting in the "granular" editor.

tfspsound, 2021-04-30:
Hey Paul. The wavetable lfo won’t effect the sample because you are using granular play mode not wavetable. Hope I’ve understood you correctly.

PaulAntioche, 2021-04-30:
sorry i was wrong ... here is the correction .
In "granular" mode with an LFO on "granular position" when there is a "note on" the sample is not played on the actual LFO position but on the position setting in the "granular" editor.

MortalJS, 2022-02-23:
Hi @PaulAntioche 
Can you check if the problem occurs in firmware 1.5?

PaulAntioche, 2022-03-07:
yes the problem persists.

PaulAntioche, 2022-03-07:
yes the problem persists