polyend/TrackerBetaTesting #1019: [1.4.0b1]Multiple files are automatically selected at sample loader.

modmodmodmod, 2021-04-30:
I linked the video.

[https://drive.google.com/file/d/19GthFgTfwlQ4Q8VLizaRPH5cmF4Vssjc/view?usp=sharing](https://drive.google.com/file/d/19GthFgTfwlQ4Q8VLizaRPH5cmF4Vssjc/view?usp=sharing)

I've never seen this bug in previous versions.
It happens every time even in another project.

PITmast, 2021-05-06:
You pressed Shift+Home (No).
Press Shift+Paste to back to normal.

Regards

modmodmodmod, 2021-05-06:
Thank you.  Sure, I usually press the shift key and home button when I would like to move cursor to the top of file browser.

But that's the official way, and of course I think it's a paste command if l press shift key and paste button as you say. 

The 1019 bug report hasn't happened so far since it was fixed by rebooting:-)

ambv, 2021-05-23:
Looks like somehow the Shift state on your keyboard locked until you reset your device.

closed 2022-02-02 by MortalJS