polyend/TrackerBetaTesting #1020: 1.3.1  Missing sample folders

Jiggg, 2021-04-30:
I've created some sample packs for personal use, using 44.1kHz 16bit for use within the tracker and other devices. The packs/folders have sub folders ie 1, 2, 3 or 105bpm, 120bpm etc to split the folders into more manageable 99 (or less) samples to meet the 100 sample folder limit, however some of the folders don't display and are hidden in the Sample directory when viewed on the Tracker.

I've checked these folders in some other gear that I own and they're visible and the samples play fine, so I suspect this is a bug related to the Tracker.

I load these folders directly onto the card through a card reader plugged into my computer and drop them straight into the Sample folder.

I use Audacity to chop up my samples.

I have had a similar problem recently of having a folder which wasn't visible in the Tracker, containing 256 files, however once I divided the files into sub folders the samples where then visible. This hasn't worked for the last 2 folders that I added which relates to me starting this bug report.

I have also had a large folder created by myself that the Tracker automatically split into 6 folders, so I don't think the quantity of samples is the issue here.

Jiggg, 2021-04-30:
Ok, i've since learned of a workaround from other users on Elektronauts.

Dropping the sample folders directly onto the card instead of in the Samples folder on the card makes the folders visible. So I guess it is still actually a bug, but with a workaround.

theorize999, 2021-05-02:
do you have more than 100 folders in your samples folder?

Jiggg, 2021-05-02:
There are 87 master folders in the Sample folder, and no idea how many sub folders there are, but there must be hundreds. Then there are the 2 folders I put into the root folder that I dropped directly onto the card.

If a fix is forthcoming, I hope the project files contain all the sample information required by the projects and I don't lose the pathways if I move them back into the Sample folder in the directory. I suspect it is, but haven't checked yet.

closed 2021-12-02 by Jacknametrouble