polyend/TrackerBetaTesting #1022: Real time song editing

tfspsound, 2021-04-30:
I don’t know if this is possible but I think it would be useful if there was an edit mode we could go into when the song is playing whereby we could select a pattern from the song as it plays and edit it without the pattern on display changing as the song progresses.

UserOverflow, 2021-04-30:
The same issue here. I would to use it as in other sequencers to change for example beat (add new notes) while playing in loop.

doelman, 2021-04-30:
Also in desperate need for this, would really be a game changer to be able
to implement ideas on the fly while jamming /performing

ketumatek, 2021-05-03:
What could be great too is the possibility to play all your sample with the grid pad in song mode.
Yet I could only play one sample I select ...

But like in pattern mode use instruments position in song mode will be very useful for jam, creation and live performance.

closed 2021-12-02 by Jacknametrouble