polyend/TrackerBetaTesting #1023: [1.3.1] Loss of entire track data when switching from perform mode to pattern

gordwithers, 2021-05-01:
This one may be hard to replicate, but will describe steps as best as I can, as it led to loss of song info.

Create a song that is mainly one pattern (on pattern 1), then  use patterns 2, 3, and 4 for alternate sections. I was using pattern 5 as a blank track, as I found that switching to a blank track sounded better than using the mutes. 
In performance mode, do various switching of patterns on the various tracks.
When all tracks don't have "Pattern 1" as current track, click on the Pattern button.
You will go to "pattern 1", but it isn't really "Pattern 1" - it is a reflection of the current mix of tracks playing. 
(eg, track 1 might have pattern 4 in it, track 2 might have pattern 2 in it, etc) 
I made some edits on one of the tracks (not realizing they were mixed up), and when I went back to performance mode, my entire first track was blank, because it had been left on a blank pattern when I'd gone into pattern mode.

bndofc, 2021-05-01:
yes! i had this happen in 1.4.0b1 also

trutypesounds, 2021-11-19:
Still have this on 1.4.0 from time to time, sometimes pattern data is missing on loading projects
@ambv  @Jacknametrouble

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS

trutypesounds, 2022-08-31:
@MortalJS its still in 1.6
dont know how exactly to reproduce, it just happens random, doing something in the pattern, go to performance mode, exit to pattern, pattern is empty with no sound, sometimes pattern data corrupts 
(pt powered via powerbank usually)