polyend/TrackerBetaTesting #1024: [1.4.1b] « Jog wheel » doesn’t respond well

ketumatek, 2021-05-01:
When I m in sample editor I want to go for example at the end of it, I use the jog wheel.

When I was on firmware 1.2 I didn't have any issue but since 1.3 I have this problem.

Problem = I have to turn the jog wheel so much that it took me a long time to arrive at the end of the sample and the same if I have to go back..it is like the jog wheel seems not working...it is very slow or seems not moving.

Before with 3 turns maw I was at the end of the sample now I can make 20 round I m at the Alf of the sample.

It happens any time on all the sample...and for all the feature...it is faster to use the navigation key...

I think it is a bug other wise it is my machine who is damage...

If you have the same problem tell me otherwise I have to send my machine back to the seller.

tfspsound, 2021-05-01:
Did some sampling earlier and modifying the start /end points was a real chore so maybe there is something to this.

ketumatek, 2021-05-02:
> Did some sampling earlier and modifying the start /end points was a real chore so maybe there is something to this. 

It seems to be a hardware problem, there is a video that explain how you can fix it.

https://m.youtube.com/watch?v=Zd_NHbw3-Mw&feature=youtu.be

tfspsound, 2021-05-02:
My jogwheel responds like the one in the video and has since I got it so I’m not sure about it being a dirty contacts issue, unless polyend are shipping them with dirty contacts which I also doubt. 
I don’t want to open up the unit because if anything else happens surely I’ll have invalidated my warranty. Polyend?

x2db, 2021-05-04:
I did the "JOG cleaning procedure", but in a week or so jog is jumping again. I'm on 1.3.1 and when I go and increase reverb level for an instrument the value jumps at around one position of the jog. So traveling from value 0 to max I have like 4-5 jumps.  I'm not sure if it is the pressure of my finger on different jog positions doing this  ? That shouldn't happen right ?  I also thought that CPU was busy slowing down the jog response , so I stopped playback and the same happens, erratic jog response.  In my opinion the jog is the most utilized part of the tracker thus it should be made with the best encoder. It is really frustrating to send unit back to manufacturer etc and cost of replacing unit for Polyend  is much higher than a cost of a better encoder.

Jacknametrouble, 2022-01-28:
This is a hardware issue, therefore we're closing it here as the Github space is intended for firmware development.
If you'd face this problem, please reach us out at https://help.polyend.com/submit_ticket for further customer support.

closed 2022-01-28 by Jacknametrouble