polyend/TrackerBetaTesting #1027: Open Source The Firmware

JamesTheHacker, 2021-05-02:
Polyend, look at all these issues and enhancements people are filing. Non of your team has addressed any of the issues, and we don't know if you're even working on them.

Open source the firmware so us smart folks can work on this. Or, address the users in the issues and let us know what's happening. Many of us here purchased the device based on the premise that Polyend were going to continue to support it. We work with the bugs and issues hoping that they'll be fixed.

Polyend, respond please.

If you agree thumbs up.

keith-macleod, 2021-05-02:
I agree with the goal, but I don't agree with your characterisation. They're a small team, they've been steadily improving the product for months, and have addressed a lot of the issues that have been raised here.

But, their time and resources are limited. Their vision of the end goal may be different from yours. There may be IP that they do not own. It's their product, and their decision, and I don't think bear-baiting is a constructive approach.

djkurzman, 2021-05-05:
The OP's reasoning is contradictory to the numerous bug fixes and feature updates that are clearly visible on this very project site. Any feature request is fair, but this entire request just comes off as an ignorant, condescending rant against Polyend. I personally see no reason for the Tracker to be open source while it is only just over a year old and is still being actively updated. If the post can't be edited to provide accurate and constructive reasons for wanting the firmware to be open source, the request should simply be removed.

JR2Media, 2021-05-08:
When was the last update?

djkurzman, 2021-05-09:
> When was the last update?

April 27 - https://github.com/polyend/TrackerIssues/releases

JR2Media, 2021-05-09:
Thanks dj! I have been checking the official site for a while but it still says 1.3.1.

closed 2021-05-14 by Jacknametrouble