polyend/TrackerBetaTesting #1029: [1.3.1/1.4.0b1] RV (Roll Volume Up) goes up to extremely loud levels if not triggered again

bndofc, 2021-05-03:
this one i'm not so sure if it's a bug or it's supposed to work this way. but if you have a RV FX on a step and there's nothing to tell it to stop (via STOP fx or another step after) it will keep going up in volume up to very scary levels, way up your master volume (at least in my case). the same step with RV will not retrigger the effect, it'll just keep going up in volume.

now i can understand if i just have to put the stop FX command right after or just another step, but i find this is pretty much useless once you start messing with the pattern play mode order in performance mode. if you use "back" and there isn't something the step before your RV fx it will keep going up in volume. again, in random play mode, this will happen much more often if it happens to trigger the RV FX, and it can go on for much longer if it doesn't find a step to cancel out the FX, leading to some very scary moments when using this perform function on patterns that have this type of roll. 

could it be made so it doesn't go past your set volume for the instrument? or at least +1 or 2db of your set instrument volume?

anemlife, 2021-05-03:
Agree, had some scary moments while playing pattern "back" in performance mode. 
And as far as I remember it has always been happening since they added this feature (not sure about the firmware version).

Hope they could make it work differently, I rarely use this Performance FX due to this issue.

x2db, 2021-05-04:
Yes the RV is pretty much useless unless you put OFF/FAD/CUT at the next step

trutypesounds, 2021-06-19:
> 
> 
> Agree, had some scary moments while playing pattern "back" in performance mode.
> And as far as I remember it has always been happening since they added this feature (not sure about the firmware version).
> 
> Hope they could make it work differently, I rarely use this Performance FX due to this issue.

just found out the same thing, reversed hat scared me pretty much

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS