polyend/TrackerBetaTesting #1037: [v1.3.1] Screen flickering

PITmast, 2021-05-06:
Hello everyone

Device is new, the firmware is v1.3.1 and I’ve not changed it yet.
It happens only on medium brightness settings. With high and low is absolutely fine (no flickering).
It’s very noticeable with included (long) usb cable and significantly less with another short usb cable.
So it is not the cable issue only as I thought initially.

Precisely the same behaviour is connected to laptop, to included charger and my power bank.

Can anybody confirm such a behaviour?

Regards
PITmaster

brucegill2, 2021-05-22:
Same here. I’m on 1.4. Will try it out with low and high setting and shorter cable shortly

PITmast, 2021-05-22:
Do not worry. The best part is to try any micro usb cable with such kind of adapter:
https://www.amazon.com/AmazonBasics-Micro-Female-USB-C-Adapter/dp/B07Q46FD9Y

No screen flickering issues at all ;)

Regards

deszczu77, 2021-09-11:
The same problem appeared with me since 1.3.1. and also in 1.4.0. The intensity of flickering is completely random. In earlier firmware there was no problem. I own PT MK2.

closed 2022-02-02 by MortalJS