polyend/TrackerBetaTesting #1040: [v1.3.1] Equalizer in Sample Editor is completely useless

PITmast, 2021-05-06:
Hi

Do you guys notice anything after applying this effect?
For example after boost (+12dB) mid frequency.

Regards
Piotr 

* How often does this bug happen? (every time, often, "it only happened once")
every time

* Does this bug happen in previous firmware versions?
I do not know

* Does this bug happen in every project?
* Yes

tfspsound, 2021-05-06:
Checkout the milestones for 1.4 this is on the list.

thebradclay, 2021-06-05:
@tfspsound has it been fixed in the current beta?

dirtysampling, 2021-07-21:
I confirm this. EQ  effect does not do anything.

MortalJS, 2022-01-17:
Fixed in 1.4
Duplicate with #1174.

closed 2022-01-17 by MortalJS

trutypesounds, 2022-01-18:
@MortalJS its not fixed, eq still works not like eq guys, please make it the right way!