polyend/TrackerBetaTesting #1045: [Firmware version] Bug title

WugWugs, 2021-05-08:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

brian3kb, 2021-05-18:
Try updating using the 16gb SD card that shipped with the Tracker. I have had this issue with previous updates and using the smaller capacity SD card resolved it for me.

brian3kb, 2021-05-20:
I think this has to do with the SD card reading implementation on the Tracker, in regular operation, the Tracker appears to be using the SDXC standard protocol which supports large SD (in theory up to 2TB - who knows whether the Trackers filesystem could address this amount of data though!).

But, when in firmware update mode, I would wager that the Tracker is using the SDHC standard, which has a hard limit of 32GB for card sizes - which would explain why smaller cards allow you to update without issue.

However, I did several earlier updates from a 128GB SD card, so, this is either a regression back to SDHC for updates (maybe for stability reasons?) - or, if the location of the update file resides on the first 32GB of the card and its allocation table, then it finds it, but if you've been using the Tracker for a while and have loaded up samples past 32GB, then its outside of the readable range of the SD reader when in update mode.

@ambv Do either of those explanations sound plausible? I've seen a few bugs about updates failing over the last month, might be worth putting a note on the official FAQ advising to only update using the supplied SD card?