polyend/TrackerBetaTesting #1050: Key navigation in Sample Loader

PITmast, 2021-05-09:
We all who spend times with Poly’s Tracker used to press Shift+Home a lot.
It will happen in Sample Loader too, it’s inevitable. Then strange things occurs with selection. BTW to go out of it press Shift+Paste.

But what about use the same shortcut to jump to the top of the list, hmm?

Regards

tfspsound, 2021-05-10:
Left arrow takes you to the top of the list, (it would be nice if Right arrow could take us to the bottom too.)
Hope I’ve understood correctly or would you like Shift + Home to work as well as Left arrow?

PITmast, 2021-05-10:
Right arrow would be very useful.
Shift+Home like left arrow because now it turns multiple selection without shift and that introduces a chaos.

I believe everyone agrees.
Regards

x2db, 2021-05-17:
... and the 48 pads are dead (doing nothin) while browsing samples right ?. I would suggest to use the "qwerty" function so hitting pad dedicated to letter  "n" would jump to the sample which name starts with "N". 

Alternatively the shift+pad would preview the sample in different pitch, as it is in the sample editor etc. 

Also the same qwerty function would be great in the project load screen.  Although Poly could fix the project list so that it is alphabetically sorted.